ghelle/MAVLinkJava #1: Mavlink Java

anrubiof, 2015-11-25:
Ghelle,
I am writing with reference to the generator you've been developing in order to implement JAVA in Mavlink System.
I do understand how to generate libraries, but i don't understand how to use them in my drone projects. I mean, how does the library communicate with the drone? How to send a message to the drone by using it?
Kind Regards
PS: If you prefer to send me an email, you could do it to anrubiof@alumnos.unex.es

ghelle, 2015-11-25:
Hello anrubiof,

Mavlink Java use Java Stream ou byte buffer to communicate.
So you must have an xbee/3DR radio plug on your PC to communicate via RS232, or socket, or other device like Android phone/tablet with xbee/3DR radio plug .
What is your environment? what do you want to do exactly?
Best regards

Guillaume Helle

anrubiof, 2015-11-25:
Guillaume,
First of all, I'm grateful to your aid. I'm working in a project about drone development for Universidad de Extremadura (Spain). We are a rookie group of people who only have JAVA knowledge, which means we don't know so much about Mavlink or Ardupilot. The part i'm involved in is about creating Mavlink messages with JAVA, and "googleing" i've gone to this website. We are using a PX4 drone with Ardupilot. What is the end of "MAVLinkJava"? How do I pass from having the XMLs I need (common and ardupilot i think) to "something" I can use to communicate to the drone.

If i'm not wrong, I have to download the files you've posted on github. Then I have to put the XMLs I want to use in a folder and write the params while invoking .bat file. After this, some folders are generated, isn't it? What can I do with those files?

As you can feel, I don't have much idea about it, so I would need a piece of advice from you. You will be included in our project credits for your code and your advice.
Thank you very much.
Ángel Rubio

ghelle, 2015-11-25:
The goal of MAVLinkJava is to have a MAVLink Java lib to communicate. You can use it on a ground station or in a drone if your drone is coded in Embedded java.
As i understand you have a drone whith PX4 board and Arducopter firmware so your goal is to have a Java ground station like MissionPlanner or qgroundcontrol.

After generation folder must be in org.mavlink.library project so the build of this project generate the jar lib to be used in your Java project with org.mavlink.util jar too.

how do you communicate with your drone? only RC command or with a groundstation too?

anrubiof, 2015-11-25:
Could you send me any example of how to generate the library with the ardupilot.xml and common.xml and how to use it to communicate with a drone? I mean that I'm not able to generate the library and I don't also know how to, in case I open the created library with eclipse and compile it, I don't know what I could do with the created .jar file. I struggle in this topic, that's why your advice is helpful.

I know that our drone has an RC receiver and FTDI UART cable, so we would like to start by learning an easy communication (maybe via commands) and later by more complex GUI systems we develop.

This make learning how to implement Mavlink in java a priority challenge. 

So, to sum up, what I need is to:

1. Learning how to generate the library, due to I don't understand the instructions in your github website.
2. Knowing how to use the jar I generate with eclipse in a project to control a drone.

Thank you and best regards.

Ángel Rubio

iamczar, 2016-11-27:
Hello Angel Rubio,

Did you manage to do this?

I am having the same dilema lol

Kind regards
Czar