ghelle/MAVLinkJava #3: MAVLink Message param_value / param_set not generated correctly

ecmnet, 2016-04-18:
Using the PX4 stack, I face some issues regarding the generated msg_param_set and msg_param_value classes. Although a float field is used for transport, params are distinguished by param_type and converted accordingly, while the generated classes always assume float. This results in wrong data transmission.

ghelle, 2016-04-19:
Hello Eike,

we don't have any problem with float : it is not a problem with little Endian/ big Endian?

Can you send me the PX4 xml file used for generation and the generated code?

Best regards

Guillaume

This message contains information that may be privileged or confidential and is the property of the Capgemini Group. It is intended only for the person to whom it is addressed. If you are not the intended recipient, you are not authorized to read, print, retain, copy, disseminate, distribute, or use this message or any part thereof. If you receive this message in error, please notify the sender immediately and delete all copies of this message.

ecmnet, 2016-04-19:
Hi Guillaume,
it's a bit weird and in deed it might be an non-conform use of data structure of the PX4 stack. I modified the generated classes (kind a quick hack) to cover at least INT/FLOAT differences, but I still face issues with DOUBLE (look at the 'switch' statement for my hack):

msg_param_set:

```
public byte[] encode() throws IOException {
  byte[] buffer = new byte[8+23];
   LittleEndianDataOutputStream dos = new LittleEndianDataOutputStream(new ByteArrayOutputStream());
  dos.writeByte((byte)0xFE);
  dos.writeByte(length & 0x00FF);
  dos.writeByte(sequence & 0x00FF);
  dos.writeByte(sysId & 0x00FF);
  dos.writeByte(componentId & 0x00FF);
  dos.writeByte(messageType & 0x00FF);
  switch(param_type) {
	case MAV_PARAM_TYPE.MAV_PARAM_TYPE_REAL32:
	case MAV_PARAM_TYPE.MAV_PARAM_TYPE_REAL64:
		dos.writeFloat(param_value);
		break;
	default:
		dos.writeInt((int)param_value);

	}
  dos.writeByte(target_system&0x00FF);
  dos.writeByte(target_component&0x00FF);
  for (int i=0; i<16; i++) {
    dos.writeByte(param_id[i]);
  }
```

Similar I do in the map_para_value message:

```
/**
 * Decode message with raw data
 */
public void decode(LittleEndianDataInputStream dis) throws IOException {
  int val = dis.readInt();
  param_count = (int)dis.readUnsignedShort()&0x00FFFF;
  param_index = (int)dis.readUnsignedShort()&0x00FFFF;
  for (int i=0; i<16; i++) {
    param_id[i] = (char)dis.readByte();
  }
  param_type = (int)dis.readUnsignedByte()&0x00FF;
  switch(param_type) {
	case MAV_PARAM_TYPE.MAV_PARAM_TYPE_REAL32:
	case MAV_PARAM_TYPE.MAV_PARAM_TYPE_REAL64:
		param_value = Float.intBitsToFloat(val);
		break;
	default:
		param_value = val;

	}
}
```

The original generated message did a `dos.writeFloat(param_value);` and  `param_value = Float.intBitsToFloat(val);` only.

common.xml states:

```
         <message id="22" name="PARAM_VALUE">
               <description>Emit the value of a onboard parameter. The inclusion of param_count and param_index in the message allows the recipient to keep track of received parameters and allows him to re-request missing parameters after a loss or timeout.</description>
               <field type="char[16]" name="param_id">Onboard parameter id, terminated by NULL if the length is less than 16 human-readable chars and WITHOUT null termination (NULL) byte if the length is exactly 16 chars - applications have to provide 16+1 bytes storage if the ID is stored as string</field>
               <field type="float" name="param_value">Onboard parameter value</field>
               <field type="uint8_t" name="param_type" enum="MAV_PARAM_TYPE">Onboard parameter type: see the MAV_PARAM_TYPE enum for supported data types.</field>
               <field type="uint16_t" name="param_count">Total number of onboard parameters</field>
               <field type="uint16_t" name="param_index">Index of this onboard parameter</field>
          </message>
          <message id="23" name="PARAM_SET">
               <description>Set a parameter value TEMPORARILY to RAM. It will be reset to default on system reboot. Send the ACTION MAV_ACTION_STORAGE_WRITE to PERMANENTLY write the RAM contents to EEPROM. IMPORTANT: The receiving component should acknowledge the new parameter value by sending a param_value message to all communication partners. This will also ensure that multiple GCS all have an up-to-date list of all parameters. If the sending GCS did not receive a PARAM_VALUE message within its timeout time, it should re-send the PARAM_SET message.</description>
               <field type="uint8_t" name="target_system">System ID</field>
               <field type="uint8_t" name="target_component">Component ID</field>
               <field type="char[16]" name="param_id">Onboard parameter id, terminated by NULL if the length is less than 16 human-readable chars and WITHOUT null termination (NULL) byte if the length is exactly 16 chars - applications have to provide 16+1 bytes storage if the ID is stored as string</field>
               <field type="float" name="param_value">Onboard parameter value</field>
               <field type="uint8_t" name="param_type" enum="MAV_PARAM_TYPE">Onboard parameter type: see the MAV_PARAM_TYPE enum for supported data types.</field>
          </message>
```

As you can see, the parameter value field is defined as float. But PX4 decodes/encodes those sent/received bytes depending on the field **param_type**.

I am not quite sure whether this would be solved using BigEndian. Did not try that yet.

By the way, one further question: Do you plan to support MAVLink 2.0?

Thank you and best regards,

Eike

https://github.com/ecmnet/MAVGCL

ecmnet, 2016-04-19:
I checked and re-run the generator with BigEndian. That was obviously wrong. A also set embedded to false (which was true before), which results in a slightly different behavior. Example:

Setting an int parameter to 1 in QGC show up as 1.4E-45 after decoding with MAVLink-Java.

ghelle, 2016-04-19:
> As you can see, the parameter value field is defined as float. But PX4 decodes/encodes those sent/received bytes depending on the field param_type . 
Ok is it a new feature in Mavlink or just for PX4? If it is just for PX4 do you think it can become a mavlink feature? 

> By the way, one further question: Do you plan to support MAVLink 2.0? 
Yes! :-) but i must go on Mavlink web site because i have stopped to read new features since a long time... 

Guillaume 

----- Mail original -----

ecmnet, 2016-04-19:
I do not know. I think it is a gap in the spec, defining a field as float and, at the same time, base the en/decoding on a second parameter like param_type. Any idea how I could integrate that into the generator?

ghelle, 2016-04-19:
I try to do that this evening : i am at work now... :-) 
Can you send me your 2 xml files : common and pix4 
and all the generated code patched 

Guillaume 

----- Mail original -----

ecmnet, 2016-04-19:
Thanks a lot. I only used common.xml.

[classes.zip](https://github.com/ghelle/MAVLinkJava/files/225628/classes.zip)
[common.xml.zip](https://github.com/ghelle/MAVLinkJava/files/225629/common.xml.zip)

ecmnet, 2016-04-22:
Closing. Its not a MAVLinkJava issue, but a PX4 stack issue

closed 2016-04-22 by ecmnet