ghelle/MAVLinkJava #8: Support for MAVLink2

anandbaskaran, 2016-10-25:
I am a user of MAVLinkJava!! Thank you so much Ghelle, I have been using this library for the past 6 months without any issues :) 

Right now, it is a good news for the community that [MAVLink 2](https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md) has been released. I am wondering is there any updates that need to be done on MAVLinkJava to support MAVLink 2.

ghelle, 2016-10-25:
Hello Anand! 

First thanks tu use it! :-) 
It'll be a pleasure to add MavLink2 update! 
Have you a link with the différences between 1.1 and 2.0? 
And the description of the new protocol 2.0? 
Sorry because i have a lot of work but i am enjoy to add 2.0 protocol! 

best regards 

Guillaume Helle 

----- Mail original -----

anandbaskaran, 2019-05-31:
Hello @ghelle. Seems like it is really high-time to introduce the support for Mavlink 2.0. Do you have any plans to implement the same? I could help you though.