ghelle/MAVLinkJava #9: Support to use MAVLINK with java

universalbeing, 2016-12-23:
Hello Ghelle,

First of all thanks for providing MAVLINK support with JAVA. It is a real lifeline.
We have been using this protocol with C so far and we would like to build some interesting GUI's. So we are moving over to JAVA / Android. I was referring to github documentation on how to build jars and include them into the project. I'm not sure i completely understood the process. Will you be able to provide any additional documentation on building and using MavlinkMessage API to read pixhawk serial data to string. RIght now i'm at a point where i can read pixhawk HEX bytes over serial port (ttyAMA0) on Raspi.

I would really appreciate any help possible.

Thanks,
Bharat

anandbaskaran, 2017-01-04:
Hello, I have created a documentation on working on this library till exporting as a JAR for my own reference. The library is working very well for me without any problems. Hope it will help you too [Generating MAVLinkJava Library Wiki.pdf](https://github.com/ghelle/MAVLinkJava/files/684716/Generating.MAVLinkJava.Library.Wiki.pdf)

ghelle, 2017-01-04:
Happy new year Anand ! 

And thanks a lot for your work! :-) 

Best regards 

Guillaume 

----- Mail original -----