ghelle/MAVLinkJava #10: Issue on latest build

anandbaskaran, 2017-09-25:
Hello,

I am trying to build the JAVA library from the XML file of PX4: [Link](https://github.com/mavlink/c_library_v1/blob/61a161e9d550df9d464f85ad47e3cf7b32d5bd94/message_definitions/common.xml)

With this new library, I cannot communicate with the drone :(

ghelle, 2017-09-25:
Can you send me the PX4 xml file and the common xml file?

anandbaskaran, 2017-09-25:
https://github.com/mavlink/c_library_v1/blob/61a161e9d550df9d464f85ad47e3cf7b32d5bd94/message_definitions/common.xml

ghelle, 2017-09-26:
Ok it is because the MAVLink protocol is 2.0
I need to implement these new features...
Thanks for the challenge : i'll try to do this quickly :-)

anandbaskaran, 2017-09-29:
Thank you @ghelle! Let me know when you are done! I will try it out as a first person :)

anandbaskaran, 2017-10-12:
Hi @ghelle! Is there any update?? I could also assist you if you need some help in development!