ghelle/MAVLinkJava #11: Using mavlink in JAVA to control Storm32c

polodelalvarado, 2018-05-24:
Hi everyone,

I'm currently doing a project to control the gimbal storm32 by means of Mavlink using JAVA language programming language.

I have the pixhawk connected to my computer through the COM port and it's working correctly, using JAVA to send the mavlink packets.

The storm32 has been connected to the pixhawk, ( following the http://ardupilot.org/copter/docs/common-storm32-gimbal.html#common-storm32-gimbal instructions), using UART port, which is connected to the telemetry2 of the pixhawk.

I am receiving the heartbeat message from the storm32 correctly, but when I send the "COMMAND_LONG" message to send the angles (which is explained in the http://ardupilot.org/dev/docs/code-overview-adding-support-for-a-new-mavlink-gimbal.html) , the gimbal doesn't move.

Do anyone know if I have to set or prepare the board before sending the "COMMAND_LONG" packet?

cesccosta, 2018-05-24:
Hi, the gimbal is full mounted and auto leveled? or only the board is connected to the pixhawk w/o gimbal?

polodelalvarado, 2018-05-25:
Hi cesccosta, thank you for answering!

I finally solved the problem setting the param7!

cesccosta, 2018-05-25:
ok :)

closed 2018-06-09 by polodelalvarado